JoelBender/bacpypes #533: Define protocolServicesSupported for my LocalDeviceObject

Weird-Banana, 2024-09-17:
Hi, I am using bacpypes to simulate a BACnet device while using a third party BACnet Explorer to view it.

As much as I am simulating a smaller number of BACnet objects, let's say 300 this seems to work fine, but whenever I simulate 1500 or more objects, the explorer will go into a buffer overflow error.

In the BACnet Explorer I see that it tries to access but does not find supported protocol for RPM, and because of this I think it's reading all objects until it goes into an overflow state instead of reading them chunk by chunk.
I checked with other real BACnet devices and they expose this property.

I wanted to try to expose my simulator with RPM supported but I run into these two errors:
`RuntimeError: protocolServicesSupported is provided by LocalDeviceObject and cannot be overridden`
`bacpypes.errors.ExecutionError: ('property', 'writeAccessDenied')`

Could you give me some direction on how to achieve this? It might not be the solution for the error encountered with this BACnet Explorer, but I wanted to give it a try.

Thanks

Weird-Banana, 2024-09-17:
Just in case, I can use bacpypes3 in case that is the more supported package.

Weird-Banana, 2024-09-17:
I was able to set this up, but now I in the BACnet Explorer I get this:
RCV OVERFLOW, in Wireshark buffer-overflow error

JoelBender, 2024-09-18:
Yes, BACpypes3 would be where the fix would go if you can reproduce the problem and I would back port it to the "legacy" version.  But first, could you try turning off segmentation?  If you can attach a PCAP file that shows the requests/responses that would be definitive.

I made a [gist](https://gist.github.com/JoelBender/e7cf8d5f9d681c9698d5191b33400fb9) with an application that acts as a server with lots of objects and another that uses the `CmdShell` but gets its configuration from a JSON file.  Once I bumped up the `max-segments-accepted` setting from the default of 16 to 1000 (it really only needed to be about 20) I could read the entire `object-list` at once.

Weird-Banana, 2024-09-18:
I will try to re-adapt the simulator with BACpypes3 then.

I think that the defect is on the BACnet Explorer side, but I wanted to also have an additional opinion.
I have uploaded the ws trace in a private git, sent you an invitation. Github does not allow me to upload it in a issue comment.

Thanks!

Weird-Banana, 2024-09-19:
I have done another wireshark trace of when I scan and explore another BACnet device.

I have noticed that the lenght of the Confirmed-REQ and Confirmed-ACK are always the same.
[image]

When I scan and explore my bacpypes simulator, these are always different.
[image]

I guess that this might be the problem of the buffer overflow.

Weird-Banana, 2024-09-19:
Nevermind, stupid error of mine.. I don't know why in the first screenshot the requests are double (2 req then 2 ack)..