JoelBender/bacpypes #536: COV stop after a while

kazemiprince, 2024-10-23:
I am testing COV(for analog input object) with Yabe and I see COV stop sending new values to Yabe and have to subscribe again to see new values.
sometimes it stop after 1-2 minutes and sometimes it stop sooner.

JoelBender, 2024-10-24:
Subscriptions have a "lifetime" parameter, Clause 13.14.1.1.4, what is the value from Yabe?  Do the notifications stop for different versions of the parameter?

kazemiprince, 2024-11-06:
its lifetime is 120 seconds.
notification stop randomly. I though it's because I run bacnet on another thread, but even if I run it on main , it stop again.
what I see is if I decrees the lifetime to 30 seconds, it will be much much better.

skehd13, 2024-11-11:
Checking the logs, it appears that `self.algorithm.triggered = True` is running late on line 55 of the service.Detect.py file. If we move that line to line 52, COV still appears to work.

Is this method correct?

bowao, 2025-01-17:
I can confirm this Issue.
Using Wireshark, you can see that bacpypes usually stops sending COV notifications at some point.
At the same time, you can see in the bacpypes Device_Object property: Active_COV_Subscriptions that the subscriptions are indeed still active. However, no more COV notifications will be sent by bacpypes.

Unfortunately, renewing the subscription does not solve the problem.
When the subscription is renewed (e.g. before lifetime expires), bacpypes responds with a COV notification after the acknowledge (BACnet standard post-subscription behavior), but no further COV notifications follow after that.

So if the lifetime is set to a very short period of time, for example 30 seconds, it looks as if the COV notifications work better, but in reality only every 30 seconds you see the response to the subscription which contains an initial COV notification and thus the current Present_Value.

Only if the subscription is actively canceled (subscription without issueConfirmedNotification and lifetime) by the subscriber and resubscribed, bacpypes will send COV notifications again until it stops again at some point.

How long it takes for bacpypes to stop sending COV notifications seems to vary. Sometimes it takes just a few minutes, sometimes a few or even many hours. But it will stop at some point.

Maybe just a coincidence, but I have the impression that the more often a value change occurs and therefore COV notifications are sent, the faster the COV notifications stop.

bowao, 2025-01-20:
After implementing the adjustment suggested by skehd13, I can no longer detect any failures in the COV notifications. Test period so far: 72 hours

Here again the changes that were made in bacpypes/service/detect.py: 

original (lines 50 to 55):

```
        # trigger it
        if trigger:
            deferred(self.algorithm._execute)
            if _debug: DetectionMonitor._debug("    - deferred: %r", self.algorithm._execute)

            self.algorithm._triggered = True

```
after adjustment (lines 50 to 55):

```
        # trigger it
        if trigger:
            self.algorithm._triggered = True
            deferred(self.algorithm._execute)
            if _debug: DetectionMonitor._debug("    - deferred: %r", self.algorithm._execute)

```

Many thanks @skehd13 for the fix

closed 2025-01-22 by JoelBender